mattjlewis/diozero #96: Add example for ESP over Wifi and Arduino/ESP over serial

egeu5, 2022-06-27:
Hello,
I'm planning to use you're awesome project to teach kids in java. In my opinion this is a nice way to intruduct into object orientated programming - I like "physical" objects.
I did not found any examples for ESP8266 over wifi or serial (or Arduino over serial). Is there any guide and I just did not found it? If not, please help me and I will add (after getting it working) a tutorial in you're docs.

Thank you!
Chris

mattjlewis, 2022-06-27:
Hi Chris, very glad you find it useful.

I'd recommend you start with the [Firmata section](https://www.diozero.com/concepts/remotecontrol.html#firmata) within the Remote Control page, apologies if this isn't obvious. Happy to provide further guidance if this still doesn't work - also any suggestions or updates to documentation would be greatly appreciated.

Regards,
Matt

egeu5, 2022-08-07:
Hi @mattjlewis , 
sorry for the late answear, I missed you're reply...
I already uploaded the firmata firmware on an arduino an tried it with another Java Lib and I could turn on and off a LED. But I have no idea how to start on you're lib.

mattjlewis, 2022-08-08:
Easy if you are using Maven, as per the [docs](https://www.diozero.com/application.html):
```
mvn archetype:generate -DarchetypeGroupId=com.diozero -DarchetypeArtifactId=diozero-application -DarchetypeVersion=1.3.3 -DgroupId=com.mycompany -DartifactId=mydiozeroapp -Dversion=1.0-SNAPSHOT
```
Then add this to the dependencies section in the created pom.xml:
```
		<dependency>
			<groupId>com.diozero</groupId>
			<artifactId>diozero-provider-firmata</artifactId>
			<version>1.3.3</version>
		</dependency>
```
The archetype creates a sample Java app that toggles an LED. Run this with app with the command line flag `-Ddiozero.firmata.serialPort` as per the [Firmata instructions](https://www.diozero.com/concepts/remotecontrol.html#firmata).