mattjlewis/diozero #98: Hardware revision is not supported

TheHeroGhost, 2022-07-22:
Hello,

If i try the test programm its working all but if i try a java programm im getting this error:

ws2811_init failed: Hardware revision is not supported
Exception in thread "main" java.lang.RuntimeException: Error initialising the WS                                                                                                                                                                                                                                             281x strip
        at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:154)
        at com.diozero.ws281xj.rpiws281x.WS281x.<init>(WS281x.java:121)
        at de.theheroghost.WS28Test.main(WS28Test.java:14)

can anyone help me please?
this is my code:

package de.theheroghost;

import com.diozero.ws281xj.LedDriverInterface;
import com.diozero.ws281xj.PixelAnimations;
import com.diozero.ws281xj.PixelColour;
import com.diozero.ws281xj.rpiws281x.WS281x;

public class WS28Test {

     public static void main(String[] args) {
         System.out.println("Test Neo starting");
         int brightness = 64;
         int num_pixels = 6;
         LedDriverInterface statusLed = new WS281x(18, brightness, num_pixels);

         int[] colours = PixelColour.RAINBOW;

         for (int i=0; i<250; i++) {
             for (int pixel=0; pixel<statusLed.getNumPixels(); pixel++) {
                 statusLed.setPixelColour(pixel, colours[(i+pixel) % colours.length]);
             }

             statusLed.render();
             PixelAnimations.delay(50);
         }
     }
}

and this is my raspi:

Hardware        : BCM2711
Revision        : d03115
Model           : Raspberry Pi 4 Model B Rev 1.5

mattjlewis, 2022-08-14:
I need to rebuild the rpi_ws2811 library - will do that as part of the 1.3.4 release. See https://github.com/jgarff/rpi_ws281x/issues/483

mattjlewis, 2022-12-16:
Missed the 1.3.4 release as needed to fix the cross compilation using podman - just committed new build that will be included when 1.3.5 is released

mattjlewis, 2023-01-19:
Should be fixed in 1.3.5

closed 2023-01-19 by mattjlewis