mattjlewis/diozero #102: PCA9685 not working

Ghozti, 2022-08-17:
Hello, i am new to diozero, i am trying to get my pca9685 HAT to work but I keep getting this error when running the jar

`Exception in thread "main" com.diozero.api.InvalidModeException: Invalid mode (Servo) for GPIO PinInfo [keyPrefix=GPIO, header=DEFAULT, deviceNumber=12, physicalPin=19, name=GPIO12, chip=-1, lineOffset=-1, modes=[PWM_OUTPUT]]
	at com.diozero.internal.spi.ServoDeviceFactoryInterface.provisionServoDevice(ServoDeviceFactoryInterface.java:53)
	at com.diozero.api.ServoDevice.<init>(ServoDevice.java:132)
	at com.diozero.api.ServoDevice$Builder.build(ServoDevice.java:110)`

My code is almost a copy/paste of the sampleapp for the PCA9685 servo test

 `    public static void main() {
        int pwm_freq = 1000;
        int pin_number = 12;
        test(pwm_freq, pin_number);
    }

    public static void test(int pwmFrequency, int gpio) {
        ServoTrim trim = ServoTrim.TOWERPRO_SG90;
        try (PCA9685 pca9685 = new PCA9685(I2CConstants.CONTROLLER_1,0x40,pwmFrequency);
             ServoDevice servo = ServoDevice.newBuilder(gpio).setDeviceFactory(pca9685).setTrim(trim).build()) {`

mattjlewis, 2022-08-17:
Apologies, I have recently fixed this in 1.3.4 which will be released soon.

mattjlewis, 2022-08-17:
1.3.4 has just been released (it may take a few hours to replicate). Could you please test and let me know.

Ghozti, 2022-08-17:
> 1.3.4 has just been released (it may take a few hours to replicate). Could you please test and let me know.

I have tested it and I am now getting this error which I assume has to do with my code:
`15:52:14.518 [main] WARN com.diozero.devices.PCA9685.createServoDevice - Specified Servo frequency (50) is different to that configured for the board (1000); this device has a common PWM frequency for all outputs
Exception in thread "main" java.lang.IllegalArgumentException: PWM value must 0..1, you requested 1.5
	at com.diozero.devices.PCA9685.setValue(PCA9685.java:353)
	at com.diozero.devices.PCA9685$PCA9685ServoOrPwmOutputDevice.setValue(PCA9685.java:464)
	at com.diozero.devices.PCA9685$PCA9685ServoOrPwmOutputDevice.<init>(PCA9685.java:434)
	at com.diozero.devices.PCA9685.createServoDevice(PCA9685.java:328)
	at com.diozero.internal.spi.ServoDeviceFactoryInterface.provisionServoDevice(ServoDeviceFactoryInterface.java:63)
	at com.diozero.api.ServoDevice.<init>(ServoDevice.java:132)
	at com.diozero.api.ServoDevice$Builder.build(ServoDevice.java:110)
	at pibotlib.lib.addons.hats.ServoDriver.test(ServoDriver.java:32)
	at pibotlib.lib.addons.hats.ServoDriver.main(ServoDriver.java:26)
`

mattjlewis, 2022-08-17:
Set `pwm_freq` to 50, i.e.

```
public static void main() {
    int pwm_freq = 50;
    int pin_number = 12;
    test(pwm_freq, pin_number);
}

public static void test(int pwmFrequency, int gpio) {
    ServoTrim trim = ServoTrim.TOWERPRO_SG90;
    try (PCA9685 pca9685 = new PCA9685(I2CConstants.CONTROLLER_1,0x40,pwmFrequency);
        ServoDevice servo = ServoDevice.newBuilder(gpio).setDeviceFactory(pca9685).setTrim(trim).build()) {
        ...
    }
}
```

Ghozti, 2022-08-17:
seems to have worked, thank you for the fix!

closed 2022-08-17 by Ghozti