mattjlewis/diozero #103: Will it be implemented: Orange Pi 3 LTS (allwinner sun50i-h6)?

glebkacorp, 2022-09-14:
I have Orange Pi 3 LTS (allwinner sun50i-h6).
I use lib ws281x-java-*
When trying to start WS2812b strip management via SPI:
"Hardware revision not supported"

OS: debian_bullseye_desktop_xfce_linux5.16.17

cat /proc/cpuinfo
Hardware    :  nothing
Revision    : nothing
Serial: nothing

/proc/device-tree/system - absent

there is only: cat /proc/device-tree/compatible
xunlong,orangepi-3-ltsallwinner,sun50i-h6

Are there plans to add device support to the library?

mattjlewis, 2022-10-04:
Should be pretty straightforward as I think that device is nearly identical to the Orange Pis already supported.

mattjlewis, 2022-10-05:
Can you share the contents of this file please:
`cat -v /proc/device-tree/model`

mattjlewis, 2022-12-16:
Added changes to support detection of this board. Can you share output of `gpiodetect` and `gpioinfo` commands please.

glebkacorp, 2022-12-19:
`gpiodetect`
result:
```
gpiochip0 [7022000.pinctrl] (64 lines)
gpiochip1 [300b000.pinctrl] (256 lines)
```

`gpioinfo`

```
gpiochip0 - 64 lines:
	line   0:      unnamed       kernel   input  active-high [used]
	line   1:      unnamed       kernel   input  active-high [used]
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed    "red-led"  output  active-high [used]
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed  "green-led"  output  active-high [used]
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       kernel   input  active-high [used]
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed      "reset"  output   active-low [used]
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       unused   input  active-high 
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused   input  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 
	line  54:      unnamed       unused   input  active-high 
	line  55:      unnamed       unused   input  active-high 
	line  56:      unnamed       unused   input  active-high 
	line  57:      unnamed       unused   input  active-high 
	line  58:      unnamed       unused   input  active-high 
[... 238 lines cut ...]
	line 232:      unnamed       unused   input  active-high 
	line 233:      unnamed       unused   input  active-high 
	line 234:      unnamed       unused   input  active-high 
	line 235:      unnamed       unused   input  active-high 
	line 236:      unnamed       unused   input  active-high 
	line 237:      unnamed       unused   input  active-high 
	line 238:      unnamed       unused   input  active-high 
	line 239:      unnamed       unused   input  active-high 
	line 240:      unnamed       unused   input  active-high 
	line 241:      unnamed       unused   input  active-high 
	line 242:      unnamed       unused   input  active-high 
	line 243:      unnamed       unused   input  active-high 
	line 244:      unnamed       unused   input  active-high 
	line 245:      unnamed       unused   input  active-high 
	line 246:      unnamed       unused   input  active-high 
	line 247:      unnamed       unused   input  active-high 
	line 248:      unnamed       unused   input  active-high 
	line 249:      unnamed       unused   input  active-high 
	line 250:      unnamed       unused   input  active-high 
	line 251:      unnamed       unused   input  active-high 
	line 252:      unnamed       unused   input  active-high 
	line 253:      unnamed       unused   input  active-high 
	line 254:      unnamed       unused   input  active-high 
	line 255:      unnamed       unused   input  active-high 
```

glebkacorp, 2022-12-19:
I was messing with the implementation in C (rpi_ws281x project).
Led-strip only works via SPI (SPI1_MOSI pin_19/GPIO_PH 5), namely

`ws2811.c`
```
//spi_fd = open("/dev/spidev0.0", O_RDWR);
spi_fd = open("/dev/spidev1.0", O_RDWR);
```

mattjlewis, 2023-01-03:
Just pushed an update that might add support for this board. Are you able to clone, build and test? Otherwise I will include in 1.3.5 as experimental / untested.

glebkacorp, 2023-01-10:
I built diozero-ws281x-java and diozera-core. I put the received JARs into the RemoteLightServer by Lars project that I had.
as a result:
```
[2023-01-10 12:38:04] [main] INFO: Board: OrangePi 3 LTS [82c00007c71471fa] (OrangePi 3 LTS) (de.lars.remotelightserver.Main.printSystemInfo:179)
[2023-01-10 12:38:04] [main] INFO: OS: Linux [aarch64] 	Java: 11.0.17 (de.lars.remotelightserver.Main.printSystemInfo:181)
[2023-01-10 12:38:05] [Server thread] INFO: [Server] Wating for connection... (de.lars.remotelightserver.server.Server.run:74)
[2023-01-10 12:38:11] [Server thread] INFO: [Server] Client connected: /192.168.13.240:62850 (de.lars.remotelightserver.server.Server.run:78)
[2023-01-10 12:38:12] [Server thread] INFO: Creating new PixelController: WS2801, 50 pixels, Pin 10 (de.lars.remotelightserver.Main.createPixelController:117)
[2023-01-10 12:38:12] [Server thread] DEBUG: Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath (com.diozero.util.LibraryLoader.loadLibrary:84)
[2023-01-10 12:38:12] [Server thread] DEBUG: Loaded library 'diozero-system-utils' from classpath (com.diozero.util.LibraryLoader.loadLibrary:93)
[2023-01-10 12:38:12] [Server thread] DEBUG: Using native device factory class DefaultDeviceFactory (com.diozero.sbc.DeviceFactoryHelper.initialise:84)
[2023-01-10 12:38:12] [Server thread] DEBUG: Using chardev GPIO implementation (com.diozero.internal.provider.builtin.DefaultDeviceFactory.start:101)
[2023-01-10 12:38:12] [Server thread] DEBUG: Found 2 GPIO chips (com.diozero.internal.provider.builtin.DefaultDeviceFactory.start:110)
[2023-01-10 12:38:12] [Server thread] TRACE: Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt' (com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition:123)
[2023-01-10 12:38:12] [Server thread] TRACE: Looking for board def file '/boarddefs/xunlong.txt' (com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition:123)
[2023-01-10 12:38:12] [Server thread] TRACE: Looking for board def file '/boarddefs/allwinner_sun50i-h6.txt' (com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition:123)
[2023-01-10 12:38:12] [Server thread] TRACE: Looking for board def file '/boarddefs/allwinner.txt' (com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition:123)
[2023-01-10 12:38:12] [Server thread] DEBUG: Memory mapped GPIO is available for this board (com.diozero.internal.provider.builtin.DefaultDeviceFactory.start:200)
[2023-01-10 12:38:12] [Server thread] TRACE: Opening /dev/spidev0.0, frequency 2400000 Hz, mode MODE_0 (com.diozero.internal.provider.builtin.spi.NativeSpiDevice.<init>:69)
[2023-01-10 12:38:12] [Server thread] TRACE: Device opened, key = Native-SPI-0-0 (com.diozero.internal.spi.AbstractDeviceFactory.deviceOpened:89)
[2023-01-10 12:38:12] [Server thread] INFO: [Server] Server stopped... (de.lars.remotelightserver.server.Server.stop:136)
[2023-01-10 12:38:12] [Server thread] ERROR: com.diozero.api.RuntimeIOException: Error in spiTransfer(), response: -1
	at com.diozero.internal.provider.builtin.spi.NativeSpiDevice.write(NativeSpiDevice.java:96)
	at com.diozero.internal.provider.builtin.spi.NativeSpiDevice.write(NativeSpiDevice.java:90)
	at com.diozero.internal.provider.builtin.DefaultNativeSpiDevice.write(DefaultNativeSpiDevice.java:76)
	at com.diozero.api.SpiDevice.write(SpiDevice.java:201)
	at com.diozero.ws281xj.spi.WS281xSpi.render(WS281xSpi.java:188)
	at com.diozero.ws281xj.spi.WS281xSpi.allOff(WS281xSpi.java:195)
	at de.lars.remotelightserver.PixelController.close(PixelController.java:52)
	at de.lars.remotelightserver.Main.closePixelController(Main.java:134)
	at de.lars.remotelightserver.server.Server.stop(Server.java:138)
	at de.lars.remotelightserver.server.Server.restart(Server.java:117)
	at de.lars.remotelightserver.server.Server$1.run(Server.java:98)
	at java.base/java.lang.Thread.run(Thread.java:829) (de.lars.remotelightserver.server.Server.run:106)

```
`TRACE: Device opened, key = Native-SPI-0-0 `
`ERROR: com.diozero.api.RuntimeIOException: Error in spiTransfer(), response: -1`

As I wrote earlier, on the Orange Pi 3LTS is required to->  open(**"/dev/spidev1.0"**, O_RDWR);   // 

**spidev_1.0_** (not SPI-0-0)

(I was doing a test version rpi_ws281x project by  jgarff  ws2811.c)

mattjlewis, 2023-01-10:
The WS281xSpi constructor allows you to specify the SPI controller and chip select:

```java
	public WS281xSpi(int controller, int chipSelect, StripType stripType, int numLeds, int brightness) {
		this(controller, chipSelect, Protocol.PROTOCOL_800KHZ, stripType, numLeds, brightness);
	}
```

It looks like you are using this [library](https://github.com/Drumber/RemoteLightServer) but I can't see where it is creating the WS281xSpi instance.

glebkacorp, 2023-01-10:
`return new WS281xSpi(**1**, 0, type, numPixels, brightness);`

NOW its OK:
```
[2023-01-10 14:27:03] [Server thread] DEBUG: Memory mapped GPIO is available for this board (com.diozero.internal.provider.builtin.DefaultDeviceFactory.start:200)
[2023-01-10 14:27:03] [Server thread] TRACE: Opening /dev/spidev1.0, frequency 2400000 Hz, mode MODE_0 (com.diozero.internal.provider.builtin.spi.NativeSpiDevice.<init>:69)
[2023-01-10 14:27:03] [Server thread] TRACE: Device opened, key = Native-SPI-1-0 (com.diozero.internal.spi.AbstractDeviceFactory.deviceOpened:89)

```
/dev/spidev1.0

Good work! 
I tried to master it, but I barely know java.

Stripe and Orange Pi 3 LTS in front of me. App is works well. There are no errors.

**Perfect! Thank you!**

mattjlewis, 2023-01-10:
Brilliant. I am just making the finishing touches to support GPIO on this board. Would you be able to run some quick tests for me?
Specifically running com.diozero.sample apps.SystemInformation and GpioReadAll. Would be great if you could also attach an LED to a few GPIOs and run the LEDTest app too.

glebkacorp, 2023-01-11:
>Would you be able to run some quick tests for me?
I'm ready.
Are the tests ready yet? Or do I need to run certain classes?

mattjlewis, 2023-01-16:
Apologies for the delay - dug out my various H3, H5 and H6 based boards to test and ended up refactoring the code for all these SoCs after reading the data sheets.

The tests are simply running the following:
* com.diozore.sampleapps.GpioReadAll
* com.diozore.sampleapps.LEDTest
* com.diozore.sampleapps.perf.GpioPerfTest

I think I'm finally good to release 1.3.5 - will do so over the next couple of days.

glebkacorp, 2023-01-17:
_com.diozore.sampleapps.GpioDetect_
>`java -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.GpioDetect `

```
gpiochip0 [7022000.pinctrl] (64 lines)
gpiochip1 [300b000.pinctrl] (256 lines)
```
***************************************
_com.diozore.sampleapps.GpioReadAll_
>`java -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.GpioReadAll`

`NOTHING`

***************************************
_com.diozore.sampleapps.LEDTest_
>`java -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.LEDTest 1`

```
11:15:42.059 [main] ERROR com.diozero.sampleapps.LEDTest.test - Error: com.diozero.api.NoSuchDeviceException: No such GPIO #1: com.diozero.api.NoSuchDeviceException: No such GPIO #1
    at com.diozero.sbc.BoardPinInfo.lambda$getByGpioNumberOrThrow$0(BoardPinInfo.java:234)
    at java.base/java.util.Optional.orElseThrow(Optional.java:408)
    at com.diozero.sbc.BoardPinInfo.getByGpioNumberOrThrow(BoardPinInfo.java:234)
    at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:142)
    at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:129)
    at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:118)
    at com.diozero.devices.LED.<init>(LED.java:82)
    at com.diozero.sampleapps.LEDTest.test(LEDTest.java:74)
    at com.diozero.sampleapps.LEDTest.main(LEDTest.java:55)
```
***************************************
_com.diozore.sampleapps.perf.GpioPerfTest_
>`java -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.perf.GpioPerfTest 12 100000`
```
Exception in thread "main" com.diozero.api.NoSuchDeviceException: No such GPIO #12
    at com.diozero.sbc.BoardPinInfo.lambda$getByGpioNumberOrThrow$0(BoardPinInfo.java:234)
    at java.base/java.util.Optional.orElseThrow(Optional.java:408)
    at com.diozero.sbc.BoardPinInfo.getByGpioNumberOrThrow(BoardPinInfo.java:234)
    at com.diozero.sampleapps.perf.GpioPerfTest.main(GpioPerfTest.java:88)
```
****and****
>`java -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar:diozero-provider-pigpio-1.3.5.jar:pigpioj-java-2.6.1.jar com.diozero.sampleapps.perf.GpioPerfTest 12 5000000`
```
jan. 17, 2023 10:37:44 AM uk.pigpioj.LibraryUtil loadLibrary
SEVERE: Error loading library from classpath '/lib/linux-aarch64/libpigpioj.so': java.lang.UnsatisfiedLinkError: /tmp/libpigpioj5948021485997311462so: libpigpio.so.1: it is impossible to open a shared object file: There is no such file or directory
Exception in thread "main" java.util.ServiceConfigurationError: com.diozero.internal.spi.NativeDeviceFactoryInterface: Provider com.diozero.internal.provider.pigpioj.PigpioJDeviceFactory could not be instantiated
    at java.base/java.util.ServiceLoader.fail(ServiceLoader.java:582)
    at java.base/java.util.ServiceLoader$ProviderImpl.newInstance(ServiceLoader.java:804)
    at java.base/java.util.ServiceLoader$ProviderImpl.get(ServiceLoader.java:722)
    at java.base/java.util.ServiceLoader$3.next(ServiceLoader.java:1395)
    at java.base/java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1812)
    at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:127)
    at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:502)
    at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:488)
    at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474)
    at java.base/java.util.stream.FindOps$FindOp.evaluateSequential(FindOps.java:150)
    at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
    at java.base/java.util.stream.ReferencePipeline.findFirst(ReferencePipeline.java:543)
    at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:80)
    at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:110)
    at com.diozero.sampleapps.perf.GpioPerfTest.main(GpioPerfTest.java:86)
Caused by: java.lang.RuntimeException: Error loading native library 'pigpioj'
    at uk.pigpioj.PigpioJNI.initialise(PigpioJNI.java:17)
    at uk.pigpioj.PigpioJNI.<init>(PigpioJNI.java:51)
    at uk.pigpioj.PigpioJ.autoDetectedImplementation(PigpioJ.java:22)
    at com.diozero.internal.provider.pigpioj.PigpioJDeviceFactory.<init>(PigpioJDeviceFactory.java:84)
    at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
    at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
    at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
    at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
    at java.base/java.util.ServiceLoader$ProviderImpl.newInstance(ServiceLoader.java:780)
    ... 13 more
```

glebkacorp, 2023-01-17:
I hope I called the classes correctly.

mattjlewis, 2023-01-17:
Thank you for this. Could you just run with these 4 JAR files on the class path please:
`tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar`

i.e. make sure not to include `diozero-provider-pigpio-1.3.5.jar:pigpioj-java-2.6.1.jar`

Also, can you turn on debugging with `-Dtinylog.level@com.diozero=trace`

e.g. 
```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.SystemInformation
```

mattjlewis, 2023-01-17:
Also I think GPIO numbers 111 and 360 would be good ones to test with for the LEDTest example.

For the pinout for this board I have followed the "[Orange Pi 3 LTS v1.2 pinout diagram](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/orange-pi-3-LTS.html)" section in the documentation.

glebkacorp, 2023-01-17:
**com.diozero.sampleapps.SystemInformation**
```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.SystemInformation

Exception in thread "main" java.lang.NoClassDefFoundError: org/fusesource/jansi/Ansi
    at com.diozero.sampleapps.SystemInformation.main(SystemInformation.java:69)
Caused by: java.lang.ClassNotFoundException: org.fusesource.jansi.Ansi
    at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
    at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
    at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
    ... 1 more
```
*************
**com.diozero.sampleapps.GpioDetect**

```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.GpioDetect

12:45:37.146 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
12:45:37.271 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
gpiochip0 [7022000.pinctrl] (64 lines)
gpiochip1 [300b000.pinctrl] (256 lines)
```
*************
**com.diozero.sampleapps.GpioReadAll**

```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.GpioReadAll

12:44:39.147 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
12:44:39.293 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
12:44:39.376 [main] DEBUG com.diozero.sbc.DeviceFactoryHelper.initialise - Using native device factory class DefaultDeviceFactory
12:44:39.383 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Using chardev GPIO implementation
12:44:39.449 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Found 2 GPIO chips
12:44:39.461 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt'
12:44:39.465 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong.txt'
12:44:39.470 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/allwinner_sun50i-h6.txt'
12:44:39.475 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/allwinner.txt'
12:44:39.525 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Memory mapped GPIO is available for this board
12:44:39.530 [main] TRACE com.diozero.internal.spi.BaseNativeDeviceFactory.close - close()
12:44:39.532 [main] TRACE com.diozero.internal.spi.AbstractDeviceFactory.close - close()
12:44:39.533 [main] TRACE com.diozero.internal.DeviceStates.closeAll - closeAll()
12:44:39.542 [main] TRACE com.diozero.internal.provider.builtin.DefaultDeviceFactory.shutdown - shutdown()
12:44:39.546 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
12:44:39.547 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
12:44:39.554 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
12:44:39.565 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - END
```
*************
**com.diozero.sampleapps.LEDTest 111**
_for LEDTest 360 - same message_
```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.LEDTest 111

12:38:09.508 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
12:38:09.645 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
12:38:09.724 [main] DEBUG com.diozero.sbc.DeviceFactoryHelper.initialise - Using native device factory class DefaultDeviceFactory
12:38:09.730 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Using chardev GPIO implementation
12:38:09.799 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Found 2 GPIO chips
12:38:09.813 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt'
12:38:09.818 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong.txt'
12:38:09.823 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/allwinner_sun50i-h6.txt'
12:38:09.827 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/allwinner.txt'
12:38:09.878 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Memory mapped GPIO is available for this board
[... cut ...]
12:38:09.958 [main] TRACE com.diozero.internal.provider.builtin.DefaultDeviceFactory.shutdown - shutdown()
12:38:09.963 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
12:38:09.966 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
12:38:09.970 [main] TRACE com.diozero.util.Diozero.shutdown - shutdown - END
12:38:09.977 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
12:38:09.983 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - Already shutdown
```

mattjlewis, 2023-01-17:
Ok - I'd named the board def file incorrectly. I've fixed in head if you are able to pull and rebuild diozero-core.

An easier interim option could be to create a folder called `boarddefs` and create a file called `xunlong_orangepi-3-lts.txt` with this content:

```
# https://linux-sunxi.org/Xunlong_Orange_Pi_3
# http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/orange-pi-3-LTS.html
# https://github.com/orangepi-xunlong/wiringOP#orange-pi-33-lts

General, DEFAULT, 1, 3v3
General, DEFAULT, 2, 5v
General, DEFAULT, 4, 5v
General, DEFAULT, 6, GND
General, DEFAULT, 9, GND
General, DEFAULT, 14, GND
General, DEFAULT, 17, 3v3
General, DEFAULT, 20, GND
General, DEFAULT, 25, GND

# GPIO, Header, GPIO#, Name, Physical Pin, Modes, Chip, Line Offset
# GPIO = (position of letter in alphabet - 1) * 32 + pin number
GPIO, DEFAULT, 122, I2C0_SDA, 3, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 122   # PD26 (122): TWI0_SDA
GPIO, DEFAULT, 121, I2C0_SCL, 5, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 121   # PD25 (121): TWI0_SCK
PWM, DEFAULT, 118, PD22, 7, 0, 0, DIGITAL_INPUT | DIGITAL_OUTPUT | PWM_OUTPUT, 1, 118 # PD22 (118): PWM0
GPIO, DEFAULT, 354, PL2, 8, DIGITAL_INPUT | DIGITAL_OUTPUT, 0, 2          # PL2 (354): UART2_RTS
GPIO, DEFAULT, 355, PL3, 10, DIGITAL_INPUT | DIGITAL_OUTPUT, 0, 3         # PL3 (355): UART2_CTS
GPIO, DEFAULT, 120, UART3_RXD, 11, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 120 # PD24 (120): UART3_RX
GPIO, DEFAULT, 114, PD18, 12, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 114      # PD18 (114)
GPIO, DEFAULT, 119, UART3_TX, 13, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 119  # PD23 (119): UART3_TX
GPIO, DEFAULT, 362, PL10, 15, DIGITAL_INPUT | DIGITAL_OUTPUT, 0, 10       # PL10 (362)
GPIO, DEFAULT, 111, PD15, 16, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 111      # PD15 (111)
GPIO, DEFAULT, 112, PD16, 18, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 112      # PD16 (112)
GPIO, DEFAULT, 229, SPI1_MOSI, 19, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 229 # PH5 (229): SPI1_MOSI
GPIO, DEFAULT, 230, SPI1_MISO, 21, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 230 # PH6 (230): SPI1_MISO
GPIO, DEFAULT, 117, PD21, 22, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 117      # PD21 (117)
GPIO, DEFAULT, 228, SPI1_SCLK, 23, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 228 # PH4 (228): SPI1_CLK
GPIO, DEFAULT, 227, SPI1_CS, 24, DIGITAL_INPUT | DIGITAL_OUTPUT, 1, 227   # PH3 (227): SPI1_CS
GPIO, DEFAULT, 360, PL8, 26, DIGITAL_INPUT | DIGITAL_OUTPUT, 0, 8         # PL8 (360)
```

glebkacorp, 2023-01-17:
Without changes  **com.diozero.sampleapps.SystemInformation**
```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.SystemInformation

Exception in thread "main" java.lang.NoClassDefFoundError: org/fusesource/jansi/Ansi
    at com.diozero.sampleapps.SystemInformation.main(SystemInformation.java:69)
Caused by: java.lang.ClassNotFoundException: org.fusesource.jansi.Ansi
    at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
    at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
    at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
    ... 1 more
```
***************
**com.diozero.sampleapps.GpioReadAll**

```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.GpioReadAll

15:05:38.965 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
15:05:39.112 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
15:05:39.198 [main] DEBUG com.diozero.sbc.DeviceFactoryHelper.initialise - Using native device factory class DefaultDeviceFactory
15:05:39.205 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Using chardev GPIO implementation
15:05:39.271 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Found 2 GPIO chips
15:05:39.282 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt'
15:05:39.288 [main] DEBUG com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Reading board defs file /boarddefs/xunlong_orangepi-3-lts.txt
15:05:39.324 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Validating BoardPinInfo against detected GPIO chip and line offsets...
15:05:39.330 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Validating detected GPIO chip and line offsets against BoardPinInfo...
15:05:39.389 [main] TRACE com.diozero.util.MmapIntBuffer.<init> - Invoking MmapBufferNative.createMmapBuffer(/dev/mem, 0x300b000, 4096)
15:05:39.394 [main] TRACE com.diozero.util.MmapIntBuffer.<init> - Invoking MmapBufferNative.createMmapBuffer(/dev/mem, 0x7022000, 4096)
15:05:39.490 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Memory mapped GPIO is available for this board
15:05:39.509 [main] TRACE com.diozero.internal.spi.BaseNativeDeviceFactory.close - close()
15:05:39.511 [main] TRACE com.diozero.internal.spi.AbstractDeviceFactory.close - close()
15:05:39.513 [main] TRACE com.diozero.internal.DeviceStates.closeAll - closeAll()
15:05:39.522 [main] TRACE com.diozero.internal.provider.builtin.DefaultDeviceFactory.shutdown - shutdown()
15:05:39.527 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
15:05:39.529 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
Exception in thread "main" java.lang.NoClassDefFoundError: org/fusesource/jansi/Ansi
    at com.diozero.sampleapps.GpioReadAll.printPins(GpioReadAll.java:83)
    at com.diozero.sampleapps.GpioReadAll.lambda$main$0(GpioReadAll.java:67)
    at java.base/java.lang.Iterable.forEach(Iterable.java:75)
    at com.diozero.sampleapps.GpioReadAll.main(GpioReadAll.java:67)
Caused by: java.lang.ClassNotFoundException: org.fusesource.jansi.Ansi
    at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
    at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
    at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
    ... 4 more
15:05:39.541 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
15:05:39.548 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - END

```
**************
_In fact, nothing is blinking. I don't understand._
**com.diozero.sampleapps.SystemInformatiLEDTest 111**
```
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar com.diozero.sampleapps.SystemInformatiLEDTest 111

15:06:28.338 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
15:06:28.476 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
15:06:28.555 [main] DEBUG com.diozero.sbc.DeviceFactoryHelper.initialise - Using native device factory class DefaultDeviceFactory
15:06:28.561 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Using chardev GPIO implementation
15:06:28.630 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Found 2 GPIO chips
15:06:28.645 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt'
15:06:28.650 [main] DEBUG com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Reading board defs file /boarddefs/xunlong_orangepi-3-lts.txt
[... cut ...]
15:11:15.773 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
15:11:15.776 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
15:11:15.778 [main] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
15:11:15.784 [main] TRACE com.diozero.util.Diozero.shutdown - shutdown - END
15:11:15.788 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
15:11:15.790 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - Already shutdown

```

mattjlewis, 2023-01-17:
Looking good. Apologies - for GpioReadAll and SystemInformation please add `jansi-2.4.0.jar` to the classpath.

glebkacorp, 2023-01-17:
**com.diozero.sampleapps.SystemInformation**
java -Dtinylog.level@com.diozero=trace -cp tinylog-api-2.5.0.jar:tinylog-impl-2.5.0.jar:diozero-core-1.3.5.jar:diozero-sampleapps-1.3.5.jar:jansi-2.4.0.jar com.diozero.sampleapps.SystemInformation

Local System Info
diozero version: 1.3.5
Operating System: ubuntu 22.04.1 LTS (Jammy Jellyfish) - aarch64
CPU Temperature: 58,80

15:29:42.579 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Looking for lib '/lib/linux-aarch64/libdiozero-system-utils.so' on classpath
15:29:42.717 [main] DEBUG com.diozero.util.LibraryLoader.loadLibrary - Loaded library 'diozero-system-utils' from classpath
15:29:42.794 [main] DEBUG com.diozero.sbc.DeviceFactoryHelper.initialise - Using native device factory class DefaultDeviceFactory
15:29:42.799 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Using chardev GPIO implementation
15:29:42.867 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Found 2 GPIO chips
15:29:42.880 [main] TRACE com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Looking for board def file '/boarddefs/xunlong_orangepi-3-lts.txt'
15:29:42.886 [main] DEBUG com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Reading board defs file /boarddefs/xunlong_orangepi-3-lts.txt
15:29:42.926 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Validating BoardPinInfo against detected GPIO chip and line offsets...
15:29:42.933 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Validating detected GPIO chip and line offsets against BoardPinInfo...
15:29:43.004 [main] TRACE com.diozero.util.MmapIntBuffer.<init> - Invoking MmapBufferNative.createMmapBuffer(/dev/mem, 0x300b000, 4096)
15:29:43.009 [main] TRACE com.diozero.util.MmapIntBuffer.<init> - Invoking MmapBufferNative.createMmapBuffer(/dev/mem, 0x7022000, 4096)
15:29:43.124 [main] DEBUG com.diozero.internal.provider.builtin.DefaultDeviceFactory.start - Memory mapped GPIO is available for this board
Detected Board Info
Device Factory: DefaultDeviceFactory
Board: Allwinner H6 OrangePi 3 LTS (RAM: 2 037 176 bytes, O/S: ubuntu 22.04.1 LTS (Jammy Jellyfish))
I2C Bus Numbers: 0, 1, 2

Header: DEFAULT
+-----+-----------+--------+----------+--------+-----------+-----+
+ GP# +      Name +  gpiod + Physical + gpiod  + Name      + GP# +
+-----+-----------+--------+----------+--------+-----------+-----+
|     |       3v3 |        |  1 || 2  |        | 5v        |     |
| 122 |  I2C0_SDA |  1:122 |  3 || 4  |        | 5v        |     |
| 121 |  I2C0_SCL |  1:121 |  5 || 6  |        | GND       |     |
| 118 |      PD22 |  1:118 |  7 || 8  |  0:2   | PL2       | 354 |
|     |       GND |        |  9 || 10 |  0:3   | PL3       | 355 |
| 120 | UART3_RXD |  1:120 | 11 || 12 |  1:114 | PD18      | 114 |
| 119 |  UART3_TX |  1:119 | 13 || 14 |        | GND       |     |
| 362 |      PL10 |  0:10  | 15 || 16 |  1:111 | PD15      | 111 |
|     |       3v3 |        | 17 || 18 |  1:112 | PD16      | 112 |
| 229 | SPI1_MOSI |  1:229 | 19 || 20 |        | GND       |     |
| 230 | SPI1_MISO |  1:230 | 21 || 22 |  1:117 | PD21      | 117 |
| 228 | SPI1_SCLK |  1:228 | 23 || 24 |  1:227 | SPI1_CS   | 227 |
|     |       GND |        | 25 || 26 |  0:8   | PL8       | 360 |
+-----+-----------+--------+----------+--------+-----------+-----+

15:29:43.632 [main] TRACE com.diozero.internal.spi.BaseNativeDeviceFactory.close - close()
15:29:43.635 [main] TRACE com.diozero.internal.spi.AbstractDeviceFactory.close - close()
15:29:43.637 [main] TRACE com.diozero.internal.DeviceStates.closeAll - closeAll()
15:29:43.653 [main] TRACE com.diozero.internal.provider.builtin.DefaultDeviceFactory.shutdown - shutdown()
15:29:43.662 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
15:29:43.664 [main] TRACE com.diozero.internal.provider.builtin.gpio.GpioChip.close - close()
15:29:43.668 [main] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
15:29:43.677 [main] TRACE com.diozero.util.Diozero.shutdown - shutdown - END
15:29:43.686 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - shutdown - START
15:29:43.697 [diozero Shutdown Handler] TRACE com.diozero.util.Diozero.shutdown - Already shutdown

glebkacorp, 2023-01-17:
There is a technique to flash the internal LED (on the board itself)?

mattjlewis, 2023-01-17:
Excellent. I don't believe there is a way to flash the board LEDs as they are used to represent power / disk access.

glebkacorp, 2023-01-18:
Are there any other tests required?

mattjlewis, 2023-01-18:
No, that's great - thank you for your support. I'm just updating the documentation now for the 1.3.5 release and will mark this board as supported.

closed 2023-01-18 by mattjlewis

glebkacorp, 2023-01-18:
Matthew, thank you!
Great job!