mattjlewis/diozero #105: NullPointerException in LocalSystemInfo when VERSION is not set in /etc/os-release

bwaldvogel, 2022-11-28:
According to the [documentation of /etc/os-release][1], the `VERSION` field is optional.

Arch Linux apparently does not set this field which causes the following `NullPointerException` in [LocalSystemInfo:268](https://github.com/mattjlewis/diozero/blob/5c0b697693ac3c5ef0e0118c5896d68d67035a59/diozero-core/src/main/java/com/diozero/sbc/LocalSystemInfo.java#L268):
```
Exception in thread "main" java.lang.NullPointerException: Cannot invoke "String.replace(java.lang.CharSequence, java.lang.CharSequence)" because the return value of "java.util.Properties.getProperty(String)" is null
        at com.diozero.sbc.LocalSystemInfo.<init>(LocalSystemInfo.java:268)
        at com.diozero.sbc.LocalSystemInfo.getInstance(LocalSystemInfo.java:244)
        at com.diozero.sbc.LocalBoardInfoUtil.initialiseLocalBoardInfo(LocalBoardInfoUtil.java:62)
        at com.diozero.sbc.LocalBoardInfoUtil.lookupLocalBoardInfo(LocalBoardInfoUtil.java:86)
        at com.diozero.util.LibraryLoader.loadSystemUtils(LibraryLoader.java:67)
        […]
```

[1]: https://www.freedesktop.org/software/systemd/man/os-release.html

bwaldvogel, 2022-11-28:
Please see PR https://github.com/mattjlewis/diozero/pull/106 for my suggested fix.

mattjlewis, 2022-11-28:
Brilliant - thank you. Will incorporate ASAP.

closed 2022-12-21 by mattjlewis