mattjlewis/diozero #125: Enhancement: Adding stepper motors to devices

EAGrahamJr, 2023-01-16:
Stepper motors would be a good enhancement for the set of available devices.

The defined interface would probably contain concepts and functions for:

- clockwise and counter-clockwise
- one step in either direction
- +/- relative angle
- acceleration/deceleration (variable speed)
- immediate stop

I think this would allow for most use cases.

(This has personal historical reasons :grin: ). There many other implementations for references and the above covers most actions in an OO way.

Preliminary thoughts are at https://github.com/EAGrahamJr/diozero/tree/add-stepper

mattjlewis, 2023-01-16:
Definitely. Someone has previously contributed a draft class for a "Wetterott SilentStepStick", however, I've not been able to test. Have just committed into the com.diozero.devices.sandpit package in diozero-core. I've held off to date as these are slightly more expensive devices.

EAGrahamJr, 2023-01-17:
Interesting - that implementation seems to drive in a single direction until disabled, which I guess makes sense in a 3D-printer, but doesn't seem "right" to me. I'm basing my implementations more on the current state of various drivers for things like the ubiquitous `28BYJ-48`.

EAGrahamJr, 2023-01-18:
I wouldn't hold up 1.3.5 for this - it's going to take a bit and I have a new I2C board coming next week that I'll want to add to the mix.

mattjlewis, 2023-01-18:
Agreed. I've ordered a couple of those 28BYJ-48 stepper motors. 1.3.5 seems good to go, as mentioned in another issue I'm just doing the last few changes to documentation.

EAGrahamJr, 2023-01-19:
Yikes - I'm not sure what stepper "flavor" I've got - they apparently come with two different gear-boxes (16:1 or 64:1), with some that aren't _quite_ exactly either. I'm pretty sure I've got the 64:1 one - I need to set up some protractors and a more accurate "dial indicator" than my LEGO pieces :grinning:

mattjlewis, 2023-01-29:
I’m still tidying up some internal refactoring, basically simplifying the way boards are automatically detected and making it easy to add support for new boards (have an OrangePi 5 on the way which should be interesting).
My 28BYJ-48 stepper motors have arrived, not had a chance to look into them yet. Any progress on your side? The next version will be 1.4.0 as I plan to do some further refactoring but shouldn’t affect any APIs - would be great to include some of this.

PR #130 (2023-02-21): Adding basic StepperMotor support with two implementations
Fixes #125

Not entirely sure if this matches with the 1.4 changes you have in mind, but this is currently working with the two types listed.

There's some holes in the implementations that need to be filled in for full implementation.
closed 2023-02-21 by mattjlewis