mattjlewis/diozero #131: Enhancement: AbstractDeviceFactory can "register" devices

EAGrahamJr, 2023-02-02:
There are several methods used with the `DeviceStates` internal class to determine whether or not devices are "opened" or not (`isDeviceOpened`, `deviceOpened`, `deviceClosed`). 

`deviceClosed` is actually invoked by the `AbstractDevice`. However, there is no "automatic" (or semi-automatic) means to utilized the _other_ two associated functions.

This issue is to track the addition of a method to the `AbstractDeviceFactory` for implementations to easily track device open/close.

EAGrahamJr, 2023-02-02:
I see this is actually managed in sub-classes of this factory - closing.

closed 2023-02-02 by EAGrahamJr

mattjlewis, 2023-02-02:
np - this internal aspect has evolved a bit over the years - doing my best to provide a completely seamless user experience and handle all of the device cleanup stuff.