mattjlewis/diozero #132: Refactor DeviceFactory interfaces to use a default registration method and Supplier

EAGrahamJr, 2023-02-02:
There's a bit of repetitious code in the various `DeviceFactoryInterface` definitions that would benefit from consolidation. This involves mostly the "open" check and the ability to prevent duplicate devices from being created.

Recommendation: create a `default registerDevice` method that wraps the actual device creation in a `Supplier` so that all downstream classes do not have to check for open devices.

EAGrahamJr, 2023-02-02:
(I've got this - it can definitely wait for 1.4)

mattjlewis, 2023-02-02:
Brilliant - look forward to seeing something here. You might have seen I did a commit not so long ago refactoring some internal bits and pieces. I should really make use of the GitHub project board for tracking backlog / ideas...

EAGrahamJr, 2023-02-02:
:grinning:

commit 21935c4b (2023-02-08): Consolidated checking for already opened devices.

Fixes #132

Basically replaces some boilerplate using functional call-backs for specifics around "actual" device creation.
closed 2023-02-08 by mattjlewis