mattjlewis/diozero #165: WS2811 in tinkerboard

SSNC-SushilBohaju, 2023-04-24:
Loaded library 'ws281xj' from classpath
ws2811_init failed: Hardware revision is not supported
Error: java.lang.RuntimeException: Error initialising the WS281x strip
java.lang.RuntimeException: Error initialising the WS281x strip
at com.diozero.ws281xj.rpiws281x.WS281x.(WS281x.java:171)
at com.diozero.ws281xj.rpiws281x.WS281x.(WS281x.java:138)
at com.mycompany.App.main(App.java:23)

A fatal error has been detected by the Java Runtime Environment:
SIGSEGV (0xb) at pc=0x92e1f4bc, pid=6680, tid=6696
JRE version: OpenJDK Runtime Environment (11.0.18+10) (build 11.0.18+10-post-Debian-1deb10u1)
Java VM: OpenJDK Server VM (11.0.18+10-post-Debian-1deb10u1, mixed mode, g1 gc, linux-)
Problematic frame:
C [libws281xj7417783119458007225.so+0x24bc] ws2811_fini+0xc
No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
An error report file with more information is saved as:
/home/linaro/mydiozeroproject/hs_err_pid6680.log
If you would like to submit a bug report, please visit:
https://bugs.debian.org/openjdk-11
The crash happened outside the Java Virtual Machine in native code.
See problematic frame for where to report the bug.

mattjlewis, 2023-04-24:
What hardware are you using? What O/S?

SSNC-SushilBohaju, 2023-04-24:
Tinkerboard R2.0 and tinker OS rockchip 3288..

EAGrahamJr, 2023-04-24:
Duplicate of #163

SSNC-SushilBohaju, 2023-04-25:
sample code using spi mode or any other material to use with spi mode ??

mattjlewis, 2023-05-30:
Yes, there is sample code for SPI Mode - see the sampleapps folder in the diozero-ws281x-java project.

closed 2023-05-30 by mattjlewis