mattjlewis/diozero #169: I2C devices are not thread-safe

EAGrahamJr, 2023-05-17:
The "default" `I2CDevice` uses _synchronized_ blocks around operations, but that implementation does not actually provide any sort of thread-safety _between_ devices - e.g. it is possible to run a multi-threaded program that uses two or more **different** I2C devices and have the I2C operations conflict with each other (the current use of _synchronized_ only works for an **instance** of a class, and using the internal delegate is likewise not helpful since that is also instantiated at the provider level).

Thread safety on other levels of operations may also need to be examined, but at this point, no other points have been identified.

EAGrahamJr, 2023-05-17:
Note: this is running on a Raspberry Pi 3.

The usual (partial) stack trace when this occurs is:
```java
com.diozero.api.RuntimeIOException: Error calling pigpioImpl.i2cWriteDevice(), response: -82
        at com.diozero.internal.provider.pigpioj.PigpioJI2CDevice.writeBytes(PigpioJI2CDevice.java:307)
        at com.diozero.api.I2CDevice.writeBytes(I2CDevice.java:481)

```
Once this happens, it takes some hardware resets on various devices and the Raspberry Pi itself.

EAGrahamJr, 2023-05-18:
A similar error occurs using the "native" library:
```java
Exception in thread "main" com.diozero.api.I2CException: Error in SMBus.writeBytes for device Native-I2C-1-0x49: -5
        at com.diozero.internal.provider.builtin.i2c.NativeI2CDeviceSMBus.writeBytes(NativeI2CDeviceSMBus.java:487)
        at com.diozero.api.I2CDevice.writeBytes(I2CDevice.java:481)

```

EAGrahamJr, 2023-05-18:
Hm. Note that I haven't been able to actually **generate** a conflict, so it appears this can be either closed or low-priority.

mattjlewis, 2023-05-30:
I initially decided that this wasn't a problem that diozero should attempt to solve and be entirely left to the developer / device owner. As you say, there's synchornization of communications on the bus itself, and not just the threads within a diozero application but across other applications. On reflections, maybe diozero should at least synchronize communications on the bus, across devices - internally manage and synchronize using some sort of communication channel instance. This could then be reused across I2C, SPI and Serial comms.

mattjlewis, 2023-05-30:
I'll add the generic internal communication channel mutex idea to the backlog. Would be useful for a multi-threaded Java app at the very least.

EAGrahamJr, 2023-08-19:
Updated information: I've been running a multi-threaded multi-I2C app for several months without any issues on a Raspberry Pi. I sincerely doubt that I've managed to completely avoid **all** collisions in that time, so it's possible the native I/O is providing blocking in that case?

EAGrahamJr, 2023-08-30:
From a StackOverflow answer:
```
I2C is a shared bus with multiple devices, which could be accessed from multiple processes as well as threads. So the Linux I2C driver code uses a mutex to manage access to each I2C bus.```

Both SMBus and the direct I2C access code in the kernel acquire locks.

Closing

closed 2023-08-30 by EAGrahamJr