mattjlewis/diozero #172: closing serial port causes close eventloop and shutdown

dafik, 2023-06-22:
I have several gpio listeners
when call serial.close()  gpio listeners are closed
INFO  com.diozero.internal.provider.builtin.gpio.GpioChip::eventLoop                           - Event loop finished

probably by this :
`raise(SIGINT); `
in native utils 
https://github.com/mattjlewis/diozero/blob/main/system-utils-native/src/main/c/com_diozero_internal_provider_builtin_serial_NativeSerialDevice.c#L381

having serial port open all time causes 100% cpu utilisation.

mattjlewis, 2023-08-17:
Will take a look.

dafik, 2024-05-28:
any progess on this ?

mattjlewis, 2024-05-28:
Apologies - not a simple fix - the current implementation settled on a blocking read to replicate behaviour similar to Java I/O. I added the SIGINT to interrupt any remaining blocking reads but clearly didn't fully think through the implications. I suspect I will need to switch to an event based read model under the hood to maintain the same exposed API in SerialDevice which will be a bit more complicated.