mattjlewis/diozero #180: BUG: Updates to board info causing errors/messages

EAGrahamJr, 2023-08-17:
I haven't checked any GPIO, but I2C is still working. These messages are ... concerning?
```
11:54:18.171 [MQTT Call: Rotomatic] WARN crackers.kobots.mqtt.KobotsMQTT.connectComplete - Connected
11:54:18.936 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.0 - line:
11:54:18.940 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.1 - line:
11:54:18.942 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.2 - line:
11:54:18.945 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.3 - line:
11:54:18.948 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.4 - line:
11:54:18.950 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.5 - line:
11:54:18.953 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.6 - line:
11:54:18.955 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.7 - line:
11:54:18.958 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.8 - line:
11:54:18.960 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.9 - line:
11:54:18.963 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.10 - line:
11:54:18.965 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.11 - line:
11:54:18.967 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: For input string: "DIGITAL_INPUT | DIGITAL_OUTPUT | PWM_OUTPUT" - line:
11:54:18.970 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: For input string: "DIGITAL_INPUT | DIGITAL_OUTPUT | PWM_OUTPUT" - line:
11:54:18.973 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.14 - line:
11:54:18.975 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.15 - line:
11:54:18.977 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.16 - line:
11:54:18.980 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.17 - line:
11:54:18.982 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: For input string: "DIGITAL_INPUT | DIGITAL_OUTPUT | PWM_OUTPUT" - line:
11:54:18.984 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: For input string: "DIGITAL_INPUT | DIGITAL_OUTPUT | PWM_OUTPUT" - line:
11:54:18.986 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.20 - line:
11:54:18.988 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.21 - line:
11:54:18.990 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.22 - line:
11:54:18.991 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.23 - line:
11:54:18.994 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.24 - line:
11:54:18.996 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.25 - line:
11:54:18.999 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.26 - line:
11:54:19.002 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.27 - line:
11:54:19.005 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.0 - line:
11:54:19.007 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.1 - line:
11:54:19.009 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.2 - line:
11:54:19.012 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.3 - line:
11:54:19.014 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.4 - line:
11:54:19.016 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.5 - line:
11:54:19.019 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.6 - line:
11:54:19.021 [main] WARN com.diozero.internal.board.GenericLinuxArmBoardInfo.loadBoardPinInfoDefinition - Illegal argument: No enum constant com.diozero.api.DeviceMode.7 - line:
```

mattjlewis, 2023-08-17:
Sorry about that, should all be fixed now. Wanted to put the modes list at the end of the line to make it easier to format. Missed a couple of minor errors - had a test case that flagged it but didn't throw an error, sigh.

EAGrahamJr, 2023-08-17:
Fixed

EAGrahamJr, 2023-08-17:
Apparently related:
```
15:44:36.869 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$2 - Configured pin chip and line offset (0-0) doesn't match that detected (0-12), line name 'GPIO12' - updating
15:44:36.872 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$2 - Configured pin chip and line offset (0-1) doesn't match that detected (0-13), line name 'GPIO13' - updating
15:44:36.874 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$2 - Configured pin chip and line offset (0-0) doesn't match that detected (0-18), line name 'GPIO18' - updating
15:44:36.876 [main] WARN com.diozero.internal.provider.builtin.DefaultDeviceFactory.lambda$start$2 - Configured pin chip and line offset (0-1) doesn't match that detected (0-19), line name 'GPIO19' - updating
```
This is for a Raspberry Pi 3. The same does **not** happen on a Pi 4.

mattjlewis, 2023-08-18:
Fixed - missed that I also needed to move the PWM defs. Knew I'd have trouble with this change...

EAGrahamJr, 2023-08-18:
No errors, warns, or other unusual things.

closed 2023-08-18 by EAGrahamJr