mattjlewis/diozero #188: Can't close and re-open MFRC522

Ramblurr, 2023-12-10:
I'm building a modular application where the modules can be re/deactivated at runtime. 

One of the modules wraps a MFRC522 device. Unlike other devices I'm wrapping (tested: Button, Led) when I open a `MFRC522` instance, and properly `device.close()` it, then try to re-open it again.. I get the following error:

```
1. Caused by com.diozero.api.DeviceAlreadyOpenedException
   Device Native-GPIO-25 is already in use

GpioDeviceFactoryInterface.java:   81  com.diozero.internal.spi.GpioDeviceFactoryInterface/provisionDigitalOutputDevice
  DigitalOutputDevice.java:  158  com.diozero.api.DigitalOutputDevice/<init>
  DigitalOutputDevice.java:  142  com.diozero.api.DigitalOutputDevice/<init>
  DigitalOutputDevice.java:  129  com.diozero.api.DigitalOutputDevice/<init>
              MFRC522.java:  321  com.diozero.devices.MFRC522/<init>
```

(the stacktrace looks a little funny because I am calling diozero from Clojure)

Ramblurr, 2023-12-11:
Thanks @EAGrahamJr ! I can confirm this patch fixes my issue. Now the MFRC522 can be opened/closed multiple times in the same process.

mattjlewis, 2023-12-11:
Brilliant, thank you. Hope to incorporate in the latest release once I’ve tested the Pi5 changes.

PR #189 (2024-01-12): Close reset pin
Fixes #188 by _also_ closing the reset pin DigitalDevice when this device is closed.
commit 557076c1 (2024-01-12): Close reset pin

Fixes #188 by _also_ closing the reset pin DigitalDevice when this device is closed.
closed 2024-01-12 by mattjlewis