mattjlewis/diozero #197: ServoTrim assumes 180 degree rotation

EAGrahamJr, 2024-04-13:
It took a while to figure out why my gear ratios were not matching real-world mappings until I realized that the servos I'm using (MG90s) move **more** than 180 degrees with an appropriate pulse-width setting, but the `ServoTrim` class assumes that a servo _cannot_ move more than 180 degrees.

Thus, `ServoTrim` should have additional constructors to cover these additional parameters (and adjust all the things accordingly).

mattjlewis, 2024-04-18:
I don't believe [this constructor](https://github.com/mattjlewis/diozero/blob/main/diozero-core/src/main/java/com/diozero/api/ServoTrim.java#L121) assumes 180 degree range of movement as per the new constructor comment:
```
	public ServoTrim(int midPulseWidthUs, int ninetyDegPulseWidthUs, int minPulseWidthUs, int maxPulseWidthUs) {
		this.midPulseWidthUs = midPulseWidthUs;
		this.ninetyDegPulseWidthUs = ninetyDegPulseWidthUs;
		this.minPulseWidthUs = minPulseWidthUs;
		this.maxPulseWidthUs = maxPulseWidthUs;
		minAngle = RangeUtil.map(minPulseWidthUs, midPulseWidthUs - ninetyDegPulseWidthUs,
				midPulseWidthUs + ninetyDegPulseWidthUs, 0, 180, false);
		maxAngle = RangeUtil.map(maxPulseWidthUs, midPulseWidthUs - ninetyDegPulseWidthUs,
				midPulseWidthUs + ninetyDegPulseWidthUs, 0, 180, false);
	}
```

EAGrahamJr, 2024-04-19:
Well, I'm an idiot -- I was putting the same value for the 90 degree as the mid-point, so of course it was coming out odd.

At least my new constructor is a little easier to use? :blush:

PR #198 (2024-04-18): Allows ServoTrim to set 0 to X arc for the motor.
Fixes #197

The default mapping would turn out non-zero based angle values because the ddefault assumption is for a 180-degree motor. This new constructor set up the servo to turn from it's baseline 0 degrees to whatever maximum is supplied.

Also added a definition for MG90S metal-gear micro-servos.
commit 41a46573 (2024-04-18): Allows ServoTrim to set 0 to X arc for the motor.

Fixes #197

The default mapping would turn out non-zero based angle values because the ddefault assumption is for a 180-degree motor. This new constructor set up the servo to turn from it's baseline 0 degrees to whatever maximum is supplied.

Also added a definition for MG90S metal-gear micro-servos.
closed 2024-04-19 by EAGrahamJr