mattjlewis/diozero #212: Current libws281xj.so library not compatible with Pi CM4 (Compute Module 4) hardware revision v1.1

chrisjarram, 2024-09-03:
Hi, as per title.  The native rpi_ws281x has been updated to include this revision but the .so in Diozero is out of date.  Grateful if an update could please be provided :)   Specific rev. being used is c03141

Thanks

PaoloTorregrosa-dc, 2024-09-18:
```bash
Exception in thread "Thread-5" java.lang.ExceptionInInitializerError
	at org.***.PadCommunication.lambda$setupButtonsThreads$0(PadCommunication.java:193)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: java.lang.IllegalStateException: Duplicate key 0 (attempted merging values com.diozero.internal.provider.builtin.gpio.GpioChip@36e9cf1b and com.diozero.internal.provider.builtin.gpio.GpioChip@74a04f98)
	at java.base/java.util.stream.Collectors.duplicateKeyException(Collectors.java:135)
	at java.base/java.util.stream.Collectors.lambda$uniqKeysMapAccumulator$1(Collectors.java:182)
	at java.base/java.util.stream.ReduceOps$3ReducingSink.accept(ReduceOps.java:169)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:113)
	at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:86)
	at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:129)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:118)
	at org.***.GPIO.<clinit>(GPIO.java:26)
	... 2 more
Exception in thread "Thread-6" java.lang.NoClassDefFoundError: Could not initialize class org.***.GPIO
	at org.***.PadCommunication.lambda$setupButtonsThreads$1(PadCommunication.java:224)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.IllegalStateException: Duplicate key 0 (attempted merging values com.diozero.internal.provider.builtin.gpio.GpioChip@36e9cf1b and com.diozero.internal.provider.builtin.gpio.GpioChip@74a04f98) [in thread "Thread-5"]
	at java.base/java.util.stream.Collectors.duplicateKeyException(Collectors.java:135)
	at java.base/java.util.stream.Collectors.lambda$uniqKeysMapAccumulator$1(Collectors.java:182)
	at java.base/java.util.stream.ReduceOps$3ReducingSink.accept(ReduceOps.java:169)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:113)
	at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:86)
	at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:129)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:118)
	at org.***.GPIO.<clinit>(GPIO.java:26)
	at org.***.PadCommunication.lambda$setupButtonsThreads$0(PadCommunication.java:193)
	... 1 more
```

Hi, Im having this issue that might be related to this on a project.

mattjlewis, 2024-09-19:
@chrisjarram leave it with me - will update the build shortly.
@PaoloTorregrosa-dc - this looks like a separate issue, can you raise a separate issue please.

mattjlewis, 2024-10-10:
Just pushed 1.4.1 with recompiled rpi_ws281x library for AArch64, please test and let me know.

mattjlewis, 2025-06-09:
Assuming fixed - closing

closed 2025-06-09 by mattjlewis