mattjlewis/diozero #215: Error initializing DigitalOutputDevice after raspberry pi os update

PaoloTorregrosa-dc, 2024-09-26:
```bash
Exception in thread "Thread-5" java.lang.ExceptionInInitializerError
	at org.***.PadCommunication.lambda$setupButtonsThreads$0(PadCommunication.java:193)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: java.lang.IllegalStateException: Duplicate key 0 (attempted merging values com.diozero.internal.provider.builtin.gpio.GpioChip@36e9cf1b and com.diozero.internal.provider.builtin.gpio.GpioChip@74a04f98)
	at java.base/java.util.stream.Collectors.duplicateKeyException(Collectors.java:135)
	at java.base/java.util.stream.Collectors.lambda$uniqKeysMapAccumulator$1(Collectors.java:182)
	at java.base/java.util.stream.ReduceOps$3ReducingSink.accept(ReduceOps.java:169)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:113)
	at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:86)
	at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:129)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:118)
	at org.***.GPIO.<clinit>(GPIO.java:26)
	... 2 more
Exception in thread "Thread-6" java.lang.NoClassDefFoundError: Could not initialize class org.***.GPIO
	at org.***.PadCommunication.lambda$setupButtonsThreads$1(PadCommunication.java:224)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.IllegalStateException: Duplicate key 0 (attempted merging values com.diozero.internal.provider.builtin.gpio.GpioChip@36e9cf1b and com.diozero.internal.provider.builtin.gpio.GpioChip@74a04f98) [in thread "Thread-5"]
	at java.base/java.util.stream.Collectors.duplicateKeyException(Collectors.java:135)
	at java.base/java.util.stream.Collectors.lambda$uniqKeysMapAccumulator$1(Collectors.java:182)
	at java.base/java.util.stream.ReduceOps$3ReducingSink.accept(ReduceOps.java:169)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.diozero.internal.provider.builtin.gpio.GpioChip.openAllChips(GpioChip.java:68)
	at com.diozero.internal.provider.builtin.DefaultDeviceFactory.start(DefaultDeviceFactory.java:113)
	at com.diozero.sbc.DeviceFactoryHelper.initialise(DeviceFactoryHelper.java:86)
	at com.diozero.sbc.DeviceFactoryHelper.getNativeDeviceFactory(DeviceFactoryHelper.java:108)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:129)
	at com.diozero.api.DigitalOutputDevice.<init>(DigitalOutputDevice.java:118)
	at org.***.GPIO.<clinit>(GPIO.java:26)
	at org.***.PadCommunication.lambda$setupButtonsThreads$0(PadCommunication.java:193)
	... 1 more
```

Hi, my team is having this problem after a recent update of raspberry pi OS.
[image]
The code lines failing are:
```java
private static final int HIGH_STATE_PIN = 17;
private static final DigitalOutputDevice gpioPin = new DigitalOutputDevice(HIGH_STATE_PIN);

```

mattjlewis, 2024-09-26:
I see your kernel version is 6.6.47. My Pi4 is running 6.1.21.

Could you share the output of `gpioinfo` please.

EAGrahamJr, 2024-09-26:
I got one and this is going to be ugly:
[6-6.txt](https://github.com/user-attachments/files/17152897/6-6.txt)

PaoloTorregrosa-dc, 2024-09-26:
```
gpiochip0 - 58 lines:
	line   0:     "ID_SDA"       unused   input  active-high 
	line   1:     "ID_SCL"       unused   input  active-high 
	line   2:      "GPIO2"       unused   input  active-high 
	line   3:      "GPIO3"       unused   input  active-high 
	line   4:      "GPIO4"       unused   input  active-high 
	line   5:      "GPIO5"       unused   input  active-high 
	line   6:      "GPIO6"       unused   input  active-high 
	line   7:      "GPIO7"       unused   input  active-high 
	line   8:      "GPIO8"       unused   input  active-high 
	line   9:      "GPIO9"       unused   input  active-high 
	line  10:     "GPIO10"       unused   input  active-high 
	line  11:     "GPIO11"       unused   input  active-high 
	line  12:     "GPIO12"       unused   input  active-high 
	line  13:     "GPIO13"       unused   input  active-high 
	line  14:     "GPIO14"       unused   input  active-high 
	line  15:     "GPIO15"       unused   input  active-high 
	line  16:     "GPIO16"       unused   input  active-high 
	line  17:     "GPIO17"       unused   input  active-high 
	line  18:     "GPIO18"       unused   input  active-high 
	line  19:     "GPIO19"       unused   input  active-high 
	line  20:     "GPIO20"       unused   input  active-high 
	line  21:     "GPIO21"       unused   input  active-high 
	line  22:     "GPIO22"       unused   input  active-high 
	line  23:     "GPIO23"       unused   input  active-high 
	line  24:     "GPIO24"       unused   input  active-high 
	line  25:     "GPIO25"       unused   input  active-high 
	line  26:     "GPIO26"       unused   input  active-high 
	line  27:     "GPIO27"       unused   input  active-high 
	line  28: "RGMII_MDIO"       unused   input  active-high 
	line  29:  "RGMIO_MDC"       unused   input  active-high 
	line  30:       "CTS0"       unused   input  active-high 
	line  31:       "RTS0"       unused   input  active-high 
	line  32:       "TXD0"       unused   input  active-high 
	line  33:       "RXD0"       unused   input  active-high 
	line  34:    "SD1_CLK"       unused   input  active-high 
	line  35:    "SD1_CMD"       unused   input  active-high 
	line  36:  "SD1_DATA0"       unused   input  active-high 
	line  37:  "SD1_DATA1"       unused   input  active-high 
	line  38:  "SD1_DATA2"       unused   input  active-high 
	line  39:  "SD1_DATA3"       unused   input  active-high 
	line  40:  "PWM0_MISO"       unused   input  active-high 
	line  41:  "PWM1_MOSI"       unused   input  active-high 
	line  42: "STATUS_LED_G_CLK" "ACT" output active-high [used]
	line  43: "SPIFLASH_CE_N" unused input active-high 
	line  44:       "SDA0"       unused   input  active-high 
	line  45:       "SCL0"       unused   input  active-high 
	line  46: "RGMII_RXCLK" unused input active-high 
	line  47: "RGMII_RXCTL" unused input active-high 
	line  48: "RGMII_RXD0"       unused   input  active-high 
	line  49: "RGMII_RXD1"       unused   input  active-high 
	line  50: "RGMII_RXD2"       unused   input  active-high 
	line  51: "RGMII_RXD3"       unused   input  active-high 
	line  52: "RGMII_TXCLK" unused input active-high 
	line  53: "RGMII_TXCTL" unused input active-high 
	line  54: "RGMII_TXD0"       unused   input  active-high 
	line  55: "RGMII_TXD1"       unused   input  active-high 
	line  56: "RGMII_TXD2"       unused   input  active-high 
	line  57: "RGMII_TXD3"       unused   input  active-high 
gpiochip1 - 8 lines:
	line   0:      "BT_ON"   "shutdown"  output  active-high [used]
	line   1:      "WL_ON"       unused  output  active-high 
	line   2: "PWR_LED_OFF" "PWR" output active-low [used]
	line   3: "GLOBAL_RESET" unused output active-high 
	line   4: "VDD_SD_IO_SEL" "vdd-sd-io" output active-high [used]
	line   5:   "CAM_GPIO" "cam1_regulator" output active-high [used]
	line   6:  "SD_PWR_ON" "regulator-sd-vcc" output active-high [used]
	line   7:    "SD_OC_N"       unused   input  active-high 
gpiochip0 - 58 lines:
[... 34 lines cut ...]
	line  34:    "SD1_CLK"       unused   input  active-high 
	line  35:    "SD1_CMD"       unused   input  active-high 
	line  36:  "SD1_DATA0"       unused   input  active-high 
	line  37:  "SD1_DATA1"       unused   input  active-high 
	line  38:  "SD1_DATA2"       unused   input  active-high 
	line  39:  "SD1_DATA3"       unused   input  active-high 
	line  40:  "PWM0_MISO"       unused   input  active-high 
	line  41:  "PWM1_MOSI"       unused   input  active-high 
	line  42: "STATUS_LED_G_CLK" "ACT" output active-high [used]
	line  43: "SPIFLASH_CE_N" unused input active-high 
	line  44:       "SDA0"       unused   input  active-high 
	line  45:       "SCL0"       unused   input  active-high 
	line  46: "RGMII_RXCLK" unused input active-high 
	line  47: "RGMII_RXCTL" unused input active-high 
	line  48: "RGMII_RXD0"       unused   input  active-high 
	line  49: "RGMII_RXD1"       unused   input  active-high 
	line  50: "RGMII_RXD2"       unused   input  active-high 
	line  51: "RGMII_RXD3"       unused   input  active-high 
	line  52: "RGMII_TXCLK" unused input active-high 
	line  53: "RGMII_TXCTL" unused input active-high 
	line  54: "RGMII_TXD0"       unused   input  active-high 
	line  55: "RGMII_TXD1"       unused   input  active-high 
	line  56: "RGMII_TXD2"       unused   input  active-high 
	line  57: "RGMII_TXD3"       unused   input  active-high 
```

EAGrahamJr, 2024-09-26:
Fudge -- it's a library update because my 6.1 Pi 4 just bit the dust as well:
```
Upgrade: raspi-utils-eeprom:arm64 (20240402-4, 20240903-1), git:arm64 (1:2.39.2-1.1, 1:2.39.5-0+deb12u1), rpicam-apps-lite:arm64 (1.5.0-2, 1.5.1-1), rpi-eeprom:arm64 (24.0-1, 26.0-1), raspi-utils-otp:arm64 (20240402-4, 20240903-1), libexpat1:arm64 (2.5.0-1, 2.5.0-1+deb12u1), libpisp1:arm64 (1.0.6-1, 1.0.7-1), python3-gpiozero:arm64 (2.0-1, 2.0.1-0+rpt1), raspi-utils-core:arm64 (20240402-4, 20240903-1), libcamera-ipa:arm64 (0.3.0+rpt20240617-1, 0.3.1+rpt20240906-1), python3-picamera2:arm64 (0.3.19-1, 0.3.21-1), libcamera0.3:arm64 (0.3.0+rpt20240617-1, 0.3.1+rpt20240906-1), libpisp-common:arm64 (1.0.6-1, 1.0.7-1), libssl3:arm64 (3.0.14-1~deb12u2, 3.0.14-1~deb12u2+rpt1), raspinfo:arm64 (20240402-4, 20240903-1), raspi-utils:arm64 (20240402-4, 20240903-1), raspi-utils-dt:arm64 (20240402-4, 20240903-1), libexpat1-dev:arm64 (2.5.0-1, 2.5.0-1+deb12u1), python3-libcamera:arm64 (0.3.0+rpt20240617-1, 0.3.1+rpt20240906-1), git-man:arm64 (1:2.39.2-1.1, 1:2.39.5-0+deb12u1), python3-v4l2:arm64 (0.3.4-1, 0.3.5-1), raspberrypi-sys-mods:arm64 (20240902, 20240911), openssl:arm64 (3.0.14-1~deb12u2, 3.0.14-1~deb12u2+rpt1), linux-libc-dev:arm64 (1:6.6.31-1+rpt1, 1:6.6.47-1+rpt1)
```
[marvin-61.txt](https://github.com/user-attachments/files/17152940/marvin-61.txt)

mattjlewis, 2024-09-26:
Ok - I can see what the problem is, should be easy to fix - it's picking up the target gpiochip0 symbolic link for gpiochip4 which of course already exists, leave it with me.

crw-rw---- 1 root gpio 254,  0 Sep 26 17:55 /dev/gpiochip0
crw-rw---- 1 root gpio 254, 10 Sep 26 17:55 /dev/gpiochip10
crw-rw---- 1 root gpio 254, 11 Sep 26 17:55 /dev/gpiochip11
crw-rw---- 1 root gpio 254, 12 Sep 26 17:55 /dev/gpiochip12
crw-rw---- 1 root gpio 254, 13 Sep 26 17:55 /dev/gpiochip13
lrwxrwxrwx 1 root root       9 Sep 26 17:55 /dev/gpiochip4 -> gpiochip0

mattjlewis, 2024-09-26:
Checking code in now. Looks to be fixed:
[image]

EAGrahamJr, 2024-09-27:
Confirm fixed:
```shell
$ uname -a
Linux marvin 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

mattjlewis, 2024-09-27:
About time I did a release ...

closed 2024-09-27 by mattjlewis

damian-barker, 2024-10-09:
Sigh. i really dont want to use Maven for my simple project.. i was running some code on a raspberry pi 3b perfectly using this library from the zip file. I updated the Pi and now it doesnt work :( Is there any chance you will provide the updated zip file with the patch?

mattjlewis, 2024-10-10:
Will release 1.4.1 today