mattjlewis/diozero #219: Enhancement Request: DHT22 (DHT11 DHT2302)

AnonymousMonkeyNo12, 2025-02-13:
I would like to use the DHT22 Thermometer and Hygrometer sensor. 

It implements it's own one-wire (1 wire) protocol.

I got it working on my rpi4 using this [guide](https://buyzero.de/blogs/news/tutorial-dht22-dht11-und-am2302-temperatursensor-feuchtigkeitsensor-am-raspberry-pi-anschliessen-und-ansteuern)

Which downloads the following repo and runs the pi code.

https://github.com/adafruit/Adafruit_Python_DHT

Now I would like to use this sensor with diozero, since I am already controlling multiple other devices through it.

I am willing to attempt to implement it myself, but any pointers would be very welcome, especially in regards to:

- How to deal with [the c files](https://github.com/adafruit/Adafruit_Python_DHT/tree/master/source) 
- How to deal with different raspberry pi versions? (There are different c files depending on the device)
- Is there already some build in stuff I can use ( I have found the one wire stuff, but it seems to be some kind of standard, contrary to the DHT22 custom protocol)

Thanks in advance and cheers.

EAGrahamJr, 2025-02-16:
Just a couple of suggestions:

1. you shouldn't have to deal with the `.c` files or JNI or anything like that: `diozero` already handles that stuff
2. the Adafruit link is to a _deprecated_ driver
3. the [newer driver](https://github.com/adafruit/Adafruit_CircuitPython_DHT) would be a better reference
4. I'm not sure what you'd use for the `Pulsio` library, so Matt might have a better notion on that
5. I'd also check in the available devices for something that looks _like_ what you've got -- that's also usually a good reference.

AnonymousMonkeyNo12, 2025-03-10:
@EAGrahamJr Thank you for your input.

I was thinking I can just drop in the C driver files here and write a simple wrapper to call them, but you are suggesting re-implementing it from scratch directly in diozero by using it's existing functions, did I understand that correctly?

Is java/diozero fast enough for the custom 1 wire protocol of the DHT22? I tried to test my understanding of the protocol by writing a quick and dirty bash script, but it was too slow to detect the edges in the signal.

I looked at the existing 1 wire stuff, but it seems to me like it is meant for the standard only. Do you have any more clues on how I would declare a custom protocol (How to tell the software to set the wire to low and start listening for a specific "header" followed by the data of a certain length)?

AnonymousMonkeyNo12, 2025-03-14:
I have noticed that there is the [Firmata protocol](https://github.com/mattjlewis/diozero/blob/main/diozero-provider-firmata/src/main/java/com/diozero/internal/provider/firmata/adapter/FirmataProtocol.java) in diozero which seems to already support the dht sensors https://github.com/firmata/protocol/blob/master/dhtsensor.md

mattjlewis, 2025-03-14:
Apologies for being away - AI Masters + work... Yes - Firmata is a good choice, I've put a lot of effort into getting that working.

AnonymousMonkeyNo12, 2025-04-12:
I finally came around to working on this again and I found that Firmata requires a Firmata-enabled microcontroller attached to the PI. But I want to attach the DHT22 directly to the PI.

Any clues on where I should start implementing this in diozero?

In case it is of any use, here is the code I wrote so far - not sure if it is correct or runs. ( https://gist.github.com/AnonymousMonkeyNo12/3e07084236b5eaf7adedbc227101b7fd )

mattjlewis, 2025-06-09:
Given the precise timing requirements for DHT11 this would most likely need to be implemented in C as concluded already. If someone provided the C code I could incorporate it into diozero (the multi-platform build tools) + bolt on a simple Java API.

mattjlewis, 2025-06-09:
FYI I've also been keeping an eye on the [foreign function and memory API](https://blog.payara.fish/java-21-foreign-function-and-memory-api) Java enhancements.

AnonymousMonkeyNo12, 2025-06-10:
@mattjlewis I linked to the c files in my first comment (which are a bit old, but work for driving the DHT11) https://github.com/adafruit/Adafruit_Python_DHT/tree/master/source

From my POV this issue can be closed as I switched to using an already supported sensor (BME280) which is a lot pricier compared to the DHT11, but at least it works out of the box. 
So please make the decision whether it would be worth adding and maintaining the DHT11. 

Thank you for your time and effort!

closed 2025-06-10 by mattjlewis